flhofer/dell_TB16_firmware #1: How to update ASM USB controller ?

druby-luke, 2024-11-26:
Hi! Thanks for all your work getting the TB16 more usable and more stable!

I Just successfully updated the cable to 26.06 but I was wondering how to update the ASM USB controller ? 

Cant seem to find the usb controller  with fwupdtool.

flhofer, 2024-11-27:
Hi @druby-luke 

Welcome :) it was a frustrating journey but I'm happy to hear that it helped someone else, too.

As for the USB hub: the controller lies on the TB-extended PCIe bus and can thus be flashed with the official ASM windows tool. However, like I wrote in the instructions, there is a tool in Linux for that (with link), too, written in C# on VScode. I had a Windows-to-go lying around so I didn't bother (I'm an Eclipse person and didn't want to adventure into new Advanced' editor territory - I only hear good, though)

If you want to try out the Linux tool, I'd be happy to hear back from you. -- I could add your experience to the instructions. Otherwise, if you don't have windows somewhere, you can use Rufus with windows iso to create a Windows-to-go disk using some Of Windows VirtualBox. 

An other option could be PCI pass-through on VBox -- i might test that in the near future.

One last thing: I just found out that Macs don't like the latest hub update. So don't flash if you are using a Mac 

Best,

Flo

flhofer, 2024-12-22:
Update: virtualbox has removed pci passthrough i n v7. So not a possibility for now

druby-luke, 2024-12-22:
Should have updated sooner, sorry about that! My approach has so far been to use a Dell laptop for everything except the cable and dock. Linux does wonders for that.

So I managed to update 8 units already for friends and community members.

Will close this issue!

closed 2024-12-22 by druby-luke