flhofer/dell_TB16_firmware #2: No Displayport on Ubuntu 24.04

did-vmonroig, 2024-12-03:
Thanks for this great repo!

I've managed to update my TB16 firmware to latest versions with `fwupdmgr` and using it with Dell XPS13 9340 (2024) with Ubuntu 24.04 as operating system. Only caveat is that Displayport and VGA shows no output. Only HDMI is working so dual monitor is not possible. I tried to install `mst_03_12_002.cab` from this repo but seems already installed with message `Specified firmware is already installed '3.12.002'`.

Any thoughs? Thanks.

flhofer, 2024-12-06:
Hi @did-vmonroig 

Did you try to restart the Dock? (unplug power, hold powerbutton, replug for a second, wait 30 sec, plug back in)

You can install the cab by using `--allow-older` to overwrite. Also, how many MST devices do you see with `get-devices`? You should see two ifs something is connected?

did-vmonroig, 2024-12-08:
Hi @flhofer, 

Thanks for your support.

This is an excerpt from output from `fwupdmgr get-devices` when both HDMI and Displayport outputs are connected:

```
├─Dell WD15/TB16/TB18 wired Dock:
│     Device ID:          2d9bd19a73af0e1160733d42c55f1d9de90ed451
│     Summary:            Multi-stream transport device
│     Current version:    3.12.02
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB, DRM_DP_AUX_DEV:0x8086)
│     GUID:               87f8cd71-1e6b-5622-94e9-0e9d11bdde33 ← MST-274
│                         c1e81f96-234e-597d-b9cb-6477d0652868 ← MST-wd15-274
│                         c0f73159-4fc0-509b-80f8-5f871f3751a8 ← MST-wd15-vmm3330-274
│                         c1e2bea5-4cf2-5f84-b965-533e8e5fea45 ← MST-tb16-274
│                         be025b25-ca5c-546c-97c6-ee2160ba489d ← MST-tb16-vmm3330-274
│                         82460515-30d3-5cdf-be4c-571340842a10 ← MST-tb18-274
│                         190e8d7c-6bcf-5fab-94e7-4f8739eb90aa ← MST-tb18-vmm3330-274
│                         6ca2b423-3cc7-509a-89fd-3329e244e11f ← MST-leaf-vmm3330-274
│                         4e4e06b6-8217-5cf0-8eff-ce5e987d531a ← MST-leaf-274
│     Device Flags:       • Updatable
│                         • Unsigned Payload
│     Device Requests:    • Message

```

I tried restarting dock and seems bricked. Luckily I have another unit and after reset nothing changed. Tried also forcing firmware upgrade with no joy:

```
$ fwupdmgr install mst_03_12_002.cab 2d9bd19a73af0e1160733d42c55f1d9de90ed451 --force --allow-older
Descomprimiendo…         [  -                                    ]
Specified firmware is already installed '3.12.002'
```

Also, VGA output does not work, not only Displayport.

Any other idea is very welcome.

flhofer, 2024-12-09:
That's curious.

It's hard to brick them, I wouldn't throw it away already. I've flashed around like 50+ times on one and I always got them back working.

Anyway, you are using DP (not mini-dp) and HDMI and see only the VMM3330? Do you have a VGA monitor? Just as a test. Fwupdmgr unfortunately doesn't work without carrier..
I know, for example, that without newest cable firmware it does have some issues.

Also, are you running as root? I don't see any sudo. Btw, you shouldn't need to specify the id for the mst. Try '--allow-reinstall' as last resort. Don't know why I didn't think about it earlier..

did-vmonroig, 2024-12-09:
After a complete power cycle and pushing power button for 45 seconds, first unit bring back to life, but with no difference in Displayport output.

Tried forcing reinstalling but seems that it failed. Screen connected via HDMI flickers and then:

```
$ sudo fwupdmgr install mst_03_12_002.cab 2d9bd19a73af0e1160733d42c55f1d9de90ed451 --allow-reinstall
Reiniciando dispositivo… [*****************                      ] Less than one minute remaining…
failed to get device before update attach: device ID 2d9bd19a73af0e1160733d42c55f1d9de90ed451 was not found
```

flhofer, 2024-12-09:
Yes, the failed message happens, the device doesn't come back up right away. But flashing seems to work in the background. (Just wait a minute before you try to power-off)

Make sure you have the updated nvm cable version. The last dock I flashed, dp would not work without.

did-vmonroig, 2024-12-09:
Tried flashing cable with Cable_26_06.bin with no changes. This is my complete output, just in case there is a flashable component I could miss out:

```
$ sudo fwupdmgr get-devices 
Dell Inc. XPS 13 9340
│
├─VEN 04F3:00 04F3:31D1:
│     Device ID:          cc9c1b47c3126a46379087db3bf18fa324f317ae
│     Summary:            Touchpad
│     Current version:    0x0006
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUID:               fadadb06-b338-5cb3-b768-a15defd32971 ← HIDRAW\VEN_04F3&DEV_31D1
│                         5e04b78f-9251-5cde-9a8d-8c4165bfe0bd ← HIDRAW\VEN_04F3&DEV_31D1&MOD_00A1
│                         876410d6-a4ff-5907-9e7e-35a82748ebd2 ← ELANTP\ICTYPE_10&MOD_00A1
│                         541f5907-b97b-5260-b293-a0028a370450 ← ELANTP\ICTYPE_10&MOD_00A1&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Core™ Ultra 7 155H:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000001f
│     Vendor:             Intel
│     GUID:               e38fa480-0c3d-5240-be48-b590e507eed9 ← CPUID\PRO_0&FAM_06&MOD_AA
│                         189c1794-cc75-56bd-b1f7-8f4af3aa3f82 ← CPUID\PRO_0&FAM_06&MOD_AA&STP_4
│     Device Flags:       • Internal device
│   
├─Dell WD15/TB16/TB18 wired Dock:
│     Device ID:          2d9bd19a73af0e1160733d42c55f1d9de90ed451
│     Summary:            Multi-stream transport device
│     Current version:    3.12.02
│     Vendor:             Synaptics (DRM_DP_AUX_DEV:0x06CB, DRM_DP_AUX_DEV:0x8086)
│     GUID:               87f8cd71-1e6b-5622-94e9-0e9d11bdde33 ← MST-274
│                         c1e81f96-234e-597d-b9cb-6477d0652868 ← MST-wd15-274
│                         c0f73159-4fc0-509b-80f8-5f871f3751a8 ← MST-wd15-vmm3330-274
│                         c1e2bea5-4cf2-5f84-b965-533e8e5fea45 ← MST-tb16-274
│                         be025b25-ca5c-546c-97c6-ee2160ba489d ← MST-tb16-vmm3330-274
│                         82460515-30d3-5cdf-be4c-571340842a10 ← MST-tb18-274
│                         190e8d7c-6bcf-5fab-94e7-4f8739eb90aa ← MST-tb18-vmm3330-274
│                         6ca2b423-3cc7-509a-89fd-3329e244e11f ← MST-leaf-vmm3330-274
│                         4e4e06b6-8217-5cf0-8eff-ce5e987d531a ← MST-leaf-274
│     Device Flags:       • Updatable
│                         • Unsigned Payload
│     Device Requests:    • Message
│   
├─Meteor Lake-P [Intel Arc Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    08
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               83527278-745c-5242-b41f-485755e7d464 ← PCI\VEN_8086&DEV_7D55
│                         a6535413-2a47-5038-8e9c-6babd8bfe622 ← PCI\VEN_8086&DEV_7D55&SUBSYS_10280C87
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─PC811 SK hynix 2048GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    61001141
│     Vendor:             SK hynix (NVME:0x1C5C)
│     Serial Number:      AYD3N03891OD23U0J
│     GUID:               22ac6c7e-bfae-5862-ba50-dd098bb656a2 ← STORAGE-DELL-112729
│                         f820c090-c9af-4d68-8171-b456cde749f7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Signed Payload
│   
├─System Firmware:
[... 112 lines cut ...]
│   
├─USB4 Retimer:
│     Device ID:          23f75b8fe7e0d7d43582bccecaf3c9a0472e72b5
│     Summary:            A physical layer protocol-aware, software-transparent extension device that forms two separate electrical link segments
│     Current version:    23.81
│     Vendor:             Intel (THUNDERBOLT:0x8087)
│     GUID:               74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
│                         f39961ab-5572-52ac-98d7-efaa6c40a5be ← TBT-80870d9c-retimer1-0:3.1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Device stages updates
│   
└─Unifying Receiver:
      Device ID:          0dc20f8ef441a8f36ef2132c52a2ede279c2c8c7
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             Logitech, Inc. (HIDRAW:0x046D, USB:0x046D)
      Install Duration:   30 seconds
      GUID:               9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:       • Updatable
                          • Supported on remote server
                          • Signed Payload
```

closed 2024-12-20 by did-vmonroig