flhofer/dell_TB16_firmware #4: Where to download firmware binaries

peacechen, 2024-12-21:
I can't find the cab/bin files online. I downloaded the Dell TB16 updater exe for Windows, but am having a heck of a time extracting it. I even ran it on a Windows box but don't see where it extracts the files to.

flhofer, 2024-12-22:
Hi!

All the firmwares are available in this repository. Choose the newest unless you're using a mac

Also, you want see the mst devices unless you connect a monitor to at least one of each mst's outputs. It is also possible that they won't show unless you update the nvm firmware first. (Issue with cable fw)

Try it out and let me know

peacechen, 2024-12-24:
I totally missed the firmware files in bin. Thanks for providing those! I was able to successfully update the dock, cable, HDMI and DP firmwares to the latest versions using your instructions. The DP output works great at 4k 60Hz, whereas it never worked before the upgrade. You're a lifesaver!

flhofer, 2024-12-24:
welcome :)

closed 2024-12-24 by flhofer