flhofer/dell_TB16_firmware #5: Power button reverse enginnering

altbert, 2025-05-24:
Hi! I'm planning to modify it so the power button works with other devices like Lenovo.
Do you think it could be possible to reverse engineer the firmware that controls the power button?
Do you know which firmware is responsible for this functionality?

flhofer, 2025-05-27:
Hi @altbert,
I didn't look into the issue that much, as I don't have a non-Dell system right now. Also, while accessing the button state with a running system may be possible, the power-on may be close to impossible. Dell Laptops have a BIOS section dedicated to the Dell-branded Dock for this purpose, giving them special access rights.

Let's help each other out on this. Do you use Linux?

flhofer, 2026-09-08:
Ok, so far I've spent some afternoon on figuring out if there is a hidden HID, or anything the like that may be accessible, but no luck.

For now I will archive this issue. If you come along new info, please feel free to reopen.

Best,
F.

altbert, 2026-10-18:
With the help of AI i've managed to get button and connector LED working for a Lenovo Thinkpad
Firmware is tailored for Lenovo only (Thinkpad only maybe) using work done [here](https://github.com/MartinezTorres/thinkpad_usb_power_button) to understand Lenovo protocol.

Turns out the PD controller (TI TPS65982) manages this specific communication with the PC. There are 2 of this controllers one for the PD in the back and the other for the main thunderbolt connector.

One of this days i will be uploading the detailed process, sadly this firmware can only be flashed directly to the eeprom (W25Q80BV) used by the PD-controller using an eeprom programmer.

### Technical summary (AI summarized)

The dock button reaches the host as a Dell USB-PD vendor message (SVID `0x413C`), which a ThinkPad ignores. I changed the laptop-facing PD controller (PC1, fw 1.2.11) so it also speaks Lenovo's button protocol (SVID `0x17EF`). The dock EC is not modified, and the Dell path is left in place (not tested on a Dell laptop).

**How it works**
- An unused SVID slot (TI `0x0451`) is reused as `0x17EF` with one mode (VDO 1).
- 334 bytes of new Cortex-M0 code sit where the PD controller's flash-update commands used to be.
- Button press: EC sends `01 83 01` → dock sends an empty Attention → ThinkPad polls status (cmd `0x10`) → dock replies with `0x0D000000` (press+release) → the laptop presses its own power button.
- Works with the laptop on, asleep (S3) or off (S5).

**Key findings**
- The ThinkPad polls status at **object position 0**. Stock firmware NAKs that and maps it to the wrong mode; fixing both was the last blocker.
- Only `01 83 01` may be forwarded. The EC sends all its events the same way, so otherwise the button release and Ethernet plug/unplug also count as presses.

**Notes**
- fwupd and Dell's updater can't write this chip, and once flashed it can only be rewritten externally.

Links: https://github.com/MartinezTorres/thinkpad_usb_power_button

flhofer, 2026-10-19:
@altbert 

Awesome work! That's great news!

I didn't think of going that deep into it as i wanted to stay focused on software type solutions. But this now opens a whole new world of possibilities; for example, the same PD controller also decides on max output power based on vendor specific polling. It would be awesome if we can remove that lock too so that all devices can negotiate up-to max power (90-100+ W).

A thing to add: i updated the PD and EC controller once via software through a BIOS based update. Maybe that's only possible due to the "always allow dell" setting so that it can access the controller via PCI bus; but it's worth looking into it.

Please, feel free to add a md file in root and the bins (original and modified) in bins for this change and submit a pull request. You should get credit for this 🙂 

In the meantime I'll check if I have the matching eeprom programming device 

Best 
F.